SC-Robotics-2023/.github #20: Optimize Drive / Hook up to gamepad controller

wluxie, 2023-01-03:


wluxie, 2023-01-07:
[Checkit](https://github.com/SC-Robotics-2021/drive_2022_23/) 
[image]

wluxie, 2023-01-09:
# TODO:
- [ ] **Resolve Odrive1 issue**
- [ ] **Implement/Design a left/right turn**
- [ ] **Create a server/client ros2 package for drive**

<br>

## Resolve Odrive1 issue
Currently, if you run what we have in drive. Odrv1 will never spin, even though it calibrated properly its not going through the python script for some reason.

```py
odrv0.axis0.requested_state = AXIS_STATE_CLOSED_LOOP_CONTROL
odrv0.axis1.requested_state = AXIS_STATE_CLOSED_LOOP_CONTROL
odrv0.axis0.controller.input_vel = int(ls_y * 10)
odrv0.axis1.controller.input_vel = int(ls_y * 10)

odrv1.axis0.requested_state = AXIS_STATE_CLOSED_LOOP_CONTROL
odrv1.axis1.requested_state = AXIS_STATE_CLOSED_LOOP_CONTROL
odrv1.axis0.controller.input_vel = int(ls_y * 10)
odrv1.axis1.controller.input_vel = int(ls_y * 10)
```
This snippet right here for example; only `odrv0` will pass and set the motors into closed loop control. Whilst it doesn't do that for `odrv1` (as seen by the setup/video we sent yesterday).

I think this has to do more with code and how it should be implemented. Previously, in our test script. We have to do something like this:
```py
odrv0.axis0.requested_state = AXIS_STATE_CLOSED_LOOP_CONTROL
odrv0.axis1.requested_state = AXIS_STATE_CLOSED_LOOP_CONTROL
odrv0.axis0.controller.input_vel = 10
odrv0.axis1.controller.input_vel = 10
if odrv_1:
    odrv1.axis0.requested_state = AXIS_STATE_CLOSED_LOOP_CONTROL
    odrv1.axis1.requested_state = AXIS_STATE_CLOSED_LOOP_CONTROL
    odrv1.axis0.controller.input_vel = 10
    odrv1.axis1.controller.input_vel = 10
```
With the function def, parameters set like so:
```py
def test_motor (odrv_0, odrv_1=None):
```
Here's the full code:
```py
def test_motor (odrv_0, odrv_1=None):
    odrv0 = odrive.find_any(serial_number=odrv_0)

    print("Enter velocity: ")
    velVal = input()

    while True:
        try:
            odrv0.axis0.requested_state = AXIS_STATE_CLOSED_LOOP_CONTROL
            odrv0.axis1.requested_state = AXIS_STATE_CLOSED_LOOP_CONTROL

            odrv0.axis0.controller.input_vel = int(velVal)
            odrv0.axis1.controller.input_vel = int(velVal)

            if odrv_1:
                odrv1 = odrive.find_any(serial_number=odrv_1)

                odrv1.axis0.requested_state = AXIS_STATE_CLOSED_LOOP_CONTROL
                odrv1.axis1.requested_state = AXIS_STATE_CLOSED_LOOP_CONTROL

                odrv1.axis0.controller.input_vel = int(velVal)
                odrv1.axis1.controller.input_vel = int(velVal)
```
Its worth investigating. And try to follow this format somewhat, see if that 1 if statement makes any difference. Haven't test this out yet but do try on Monday.

<br>

## Implement/Design a left/right turn
Right now we only have the forward/backward motion implemented. So we need to work on the turning aspect of the drive. This is pretty straightforward. 
Our rover does not have in-phase steering, so all 4 motors would face in the same direction. So to turn:
You can simply put 1 side of the rover, to spin at a higher speed than the other side, creating a drift effect, thereby "turning" it.

[image]

Draft ^ (idk I literally drew this as I'm eating). Let's say your `odrv0` is the left half of your rover, and `odrv1` is the right half. By turning `odrv0` at say, `15 m/s`, and `odrv1` is only at `7.5 m/s`. The high speed on the left side will probably cause the rover to steer more into the right side. So effectively, if you want to turn left, increase the speed of the left side, and vice versa.

And I want to be able to do this motion through the right joystick of the controller. You can set this through `rs_x`. Cameron knows how it works so you can ask him, but effectively:
- If the right joystick is pushed to the right `rs_x` checks for analog input for the x-axis of it so `rs_x` is set to 1 now
```
[ _      1     _ ]
[ -1    0    *1* ]  <------ X-axis with your joystick pushed to the right
[ _     -1     _ ]
```
So TL;DR:
```py
if rs_x > 0:
    do right turn
    so odrv0 go brrrr and odrv1 go bruhbruhbruh
if rs_x < 0:
    d0 right turn
    odrv0 bruhrbruhbruh odrv1 brrrrrrrr
```

<br>

 ## Create a server/client ros2 package for drive
I forgot ros2 so ¯\_(ツ)_/¯
Back to ros2 documentation :DDDDDDDDDD

But I think we need to create a server & client package. Where the server requests a package (in our case `loki` or any pi that is hooked up to drive). And the client, sends the package, which are just analog signal from the controller to the server. 

Here's the [doc](https://docs.ros.org/en/foxy/Tutorials/Beginner-Client-Libraries/Writing-A-Simple-Py-Service-And-Client.html)

And here's a img from the doc so it looks like I know what I'm typing:
[image]

Here's a baby meme :DDD
[image]

Have fun kthxbyeeeee see you at 4pm

MickeyMauser, 2023-01-10:
[image]

CONTROL SCHEME THEORY:

— This design approach will allow the user to accurately and intuitively control the drive train of the ROVER, while implementing features that prevent user error in a way that is not cumbersome to manage.

BASIC CONTROL SCHEME DRAFT:

L2 : Clutch

- Must be engaged to control speed, turns, and set cruise control
- Must be held down to be considered engaged
- Must be engaged for turn in place
- IF CLUTCH NOT ENGAGED NO DRIVE OR TURN FUNCTIONS ARE OPERATIONAL: This serves to protect against errors in user input
- If clutch engaged, cruise control MAY be activated

Right Thumbstick: Steering

- Clutch must be engaged to function EVEN WHILE CRUISE IS ACTIVATED
- Controls turn in intended direction
- Right = Right Turn (Left side slowed, stopped, or reversed relative to current ROVER speed)
- Left = Left Turn (Right side slowed, stopped, or reversed relative to current ROVER speed)
- Can turn from 0 rps while clutch engaged (Opposite side is reversed in order to turn in place)
- Operational only on X axis

Left Thumbstick: Speed — forward/backward

- Increments speed of both left and right sides equally
- Needs clutch engaged to function
- Down = Reverse
- Up = Forward
- ***Max forward speed > Max reverse speed ***
- Speed can be saved with Cruise function
- Operational only on Y axis

L1: Cruise Control

- Requires clutch to be engaged and greater than zero value on Y axis of left thumbstick
- Saves current left thumbstick input
- Cruise CANNOT be activated while in reverse
- Cruise will NOT save right thumbstick input/right thumbstick input will not overwrite cruise values
- Must engage clutch to turn with right thumbstick
- Left Thumbstick NOT OPERATIONAL while cruise toggled on
- HAT (directional pad) can increment/decrement speed
- Cruise control ends by engaging clutch and toggling cruise

HAT (Directional PAD): Increment max speed

- Increment/decrement the maximum speed with Up and Down arrows
- Left d-pad activates optimized lower speed preset for precise movement
- Right d-pad activates optimized high speed preset just below maximum speed
- D-pad is only way to control speed while in cruise control

wluxie, 2023-01-10:
Fixed Task/Issue 1 & 2
[image]