SC-Robotics-2023/.github #9: Setup ODrives NEOs (4 motors)

wluxie, 2023-01-03:


fngld, 2023-01-04:
### 1.3.23 Drive Meeting

**ODrive Setup:**
- ODrives 2 and 4
- ODrives sitting in enclosure

**Motor Setup:**
- Odrive 4: Left middle, left back
- ODrive 2: Right back, Left front
- Face up, sitting directly on tabletop
- Encoder wires shielded, touching power wires

**Minutes**

-  Ran full calibration sequence, then ran both axes twice in a row, no errors
 -- All 4 motors appeared to run at the same velocity, as intended

- SSH into drive pi over 2.4 GHz
- 1st try: 
- - 1 neo pulled hall polarity error, someone possibly moved it during calibration
- 2nd try:
- -  same motor threw HALL POLARITY error, no one touched it
- 3rd try:
- - Removed power, reattached all wires to the same ports, straightened out wiring so that no wires touch
-  - 3 out of 4 successfully calibrated, same motor had HALL POLARITY error
- 4th try:
-  - Error throwing motor (Left Front on ODrive 4) was held in the air during calibration, hall polarity error again
- 5th try:
- - Error prone motor and its wires were placed on an insulator (2x4), threw HALL PHASE error
- 6th try:
- - All motors rewired. ODrive 
- - All motors passed calibration
- 7th try:
- - Motors returned to original configuration
-  - Error prone motor threw HALL POLARITY error
- 8th try:
- - ODrive 4 motors swapped back to the working configuration
- - ODrive 4 removed, different (unlabeled) ODrive swapped in
- - ODrive OUT OF RANGE ERROR

End of meeting

wluxie, 2023-01-07:
Solved

closed 2023-01-07 by wluxie